theelims/ESP32-sveltekit #5: ESP32-sveltekit is crashing with latest platform 6.4.0

EvEggelen, 2023-09-11:
Problem. Code builds with both 6.4.0 ( platform = espressif32@6.4.0 )and 6.3.2 ( platform = espressif32@6.3.2 ). But crashes during startup when platform 6.4.0 is used. 

*ESP32-sveltekit:**
https://github.com/theelims/ESP32-sveltekit/releases/tag/v0.2.0

**Added to platformio.ini**
[env:nodemcu-32s]
board = nodemcu-32s
board_build.partitions = min_spiffs.csv

**Hardware:**
https://www.aliexpress.com/item/1005001636295529.html

**Log when using 6.4.0**
esptool.py v4.5.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/nodemcu-32s/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:a7:32:28:a6:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00164fff...
Compressed 17536 bytes to 12203...
Writing at 0x00001000... (100 %)
Wrote 17536 bytes (12203 compressed) at 0x00001000 in 0.4 seconds (effective 358.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 1073.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 1635.5 kbit/s)...
Hash of data verified.
Compressed 1394048 bytes to 925247...
Writing at 0x00010000... (1 %)
Writing at 0x00014c6e... (3 %)
Writing at 0x00018c8f... (5 %)
Writing at 0x0001cd63... (7 %)
Writing at 0x00020d5d... (8 %)
Writing at 0x00024d60... (10 %)
Writing at 0x00028d58... (12 %)
Writing at 0x0002cd53... (14 %)
Writing at 0x00030d4e... (15 %)
Writing at 0x00035ea0... (17 %)
Writing at 0x00039e87... (19 %)
Writing at 0x00044150... (21 %)
Writing at 0x0004fd11... (22 %)
Writing at 0x0005b3fe... (24 %)
Writing at 0x0006686a... (26 %)
Writing at 0x00077ce5... (28 %)
[... 191 lines cut ...]
Leaving...
Hard resetting via RTS pin...

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[     8][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
[    45][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Running Firmware Version: 0.2.0
Starting software access point
[   288][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   394][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   395][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
[   399][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[   408][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[   421][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[   987][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[   988][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
[   988][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   995][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
Starting captive portal on 192.168.4.1

theelims, 2023-09-11:
Hi,

I've observed this behavior as well. That's why I pushed a small update 4 days ago, if you use the latest version. For now the only way around that is to stick with:
```platform = espressif32 @ 6.3.2```

It lock's up during start-up as one thread consumes all CPU power to the point, that not even the IDLE-Task is served anymore. But I have no clue what could cause this.

theelims, 2023-09-11:
It looks like esp8266-react from which it forks is affected as well. There is an issue open at https://github.com/rjwats/esp8266-react/issues/329 

I'll have a look at the solution and give it a try.

theelims, 2023-09-11:
Released v0.2.1 with the fixes from the open esp8266-react PR.

closed 2023-09-11 by theelims