theelims/ESP32-sveltekit #8: lost connection to webserver

EvEggelen, 2023-09-14:
**To reproduce**
1. Clean build and upload SW.
2. Connect to device ( 192.168.4.1)
3. Login screen shows
4. Do a refresh in browser.
5. You will get an error message in the log window and the connection is lost forever.
Sometimes I see the error something like" Too many messages in the queue" 

**ESP32-sveltekit:**
https://github.com/theelims/ESP32-sveltekit/releases/tag/v0.2.1

**Added to platformio.ini**
[env:nodemcu-32s]
board = nodemcu-32s
board_build.partitions = min_spiffs.csv

**Hardware:**
https://www.aliexpress.com/item/1005001636295529.html

**[features]**
build_flags = 
  -D FT_SECURITY=1
  -D FT_MQTT=1
  -D FT_NTP=1
  -D FT_OTA=0               ; Not recommended, will be deprectiated in an upcomming release
  -D FT_UPLOAD_FIRMWARE=1
  -D FT_DOWNLOAD_FIRMWARE=1 ; requires FT_NTP=1
  -D FT_SLEEP=1
  -D FT_BATTERY=0

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
␛[33m=> 0x40078000: ?? ??:0␛[0m
ho 0 tail 12 room 4
load:0x40080400,len:2908
␛[33m=> 0x40080400: _init at ??:?␛[0m
entry 0x400805c4
␛[33m=> 0x400805c4: ?? ??:0␛[0m
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    90][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   189][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   190][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: [ - 191_ST][W
iG[neri92][E:34iF Gendrico_even042]()event Sloppck(
: esp_wifi_set_ps failed
[   199][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 3 - STA_STOP
Running Firmware Version: 0.2.1
Starting software access point
[   439][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   447][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   448][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
[   455][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[   462][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[   474][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[  1138][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[  1138][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STO 1139][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[  1145][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
Starting captive portal on 192.168.4.1
[ 14808][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: 44:78:3e:f6:b0:21, AID: 1
[ 14808][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 15063][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[ 15064][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 32654][W][AsyncTCP.cpp:976] _poll(): pcb is NULL
New client connected to Event Source: 1 Clients connected
[ 38216][W][AsyncTCP.cpp:976] _poll(): pcb is NULL

theelims, 2023-09-14:
I'll try to reproduce this bug as soon as I'm in front of a dev board.

theelims, 2023-09-17:
I've tried to reproduce this behaviour on a ESP32 board and an ESP32-S3 board. Both refresh the login page as expected. I'm unable to reproduce this bug.

Can you give me some more details on how it is failing? What does the log exactly show when the server is not refreshing?

EvEggelen, 2023-09-19:
I tried to reproduce today. But I do not get consistent results. I will investigate further. Let you know.

I do get warning during build. Is that "normal", or is that a build problem on my side ?

 *  Executing task in folder ESP32-sveltekit-0.2.1: platformio run --target upload --target monitor --environment nodemcu-32s 

Processing nodemcu-32s (board: nodemcu-32s; framework: arduino; platform: espressif32)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
['upload']
Building interface with npm
npm install

up to date, audited 294 packages in 987ms

67 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
npm run build

> ESP32-Sveltekit Template@0.2.0 build
> vite build

vite v4.4.8 building SSR bundle for production...
transforming...
9:12:31 PM [vite-plugin-svelte] /src/routes/+page.svelte:3:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
1: <script lang="ts">import logo from "$lib/assets/logo.png";
2: import { notifications } from "$lib/components/toasts/notifications";
3: export let data;
              ^
4: </script>
5: 
9:12:31 PM [vite-plugin-svelte] /src/routes/system/update/+page.svelte:6:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
4: import { user } from "$lib/stores/user";
5: import { page } from "$app/stores";
6: export let data;
              ^
7: </script>
8: 
9:12:31 PM [vite-plugin-svelte] /src/routes/wifi/+page.svelte:3:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
1: <script lang="ts">import Accesspoint from "./Accesspoint.svelte";
2: import Wifi from "./Wifi.svelte";
3: export let data;
              ^
4: </script>
5: 
9:12:31 PM [vite-plugin-svelte] /src/routes/demo/+page.svelte:2:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
1: <script lang="ts">import Demo from "./Demo.svelte";
2: export let data;
              ^
3: </script>
4: 
9:12:31 PM [vite-plugin-svelte] /src/routes/user/+page.svelte:20:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
18: import Cancel from "~icons/tabler/x";
19: import Check from "~icons/tabler/check";
20: export let data;
               ^
21: let securitySettings;
22: async function getSecuritySettings() {
9:12:31 PM [vite-plugin-svelte] /src/routes/connections/mqtt/+page.svelte:3:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
1: <script lang="ts">import MQTT from "./MQTT.svelte";
2: import MqttConfig from "./MQTTConfig.svelte";
3: export let data;
              ^
4: </script>
5: 
9:12:31 PM [vite-plugin-svelte] /src/routes/connections/ntp/+page.svelte:2:11 Page has unused export property 'data'. If it is for external reference only, please consider using `export const data`
1: <script lang="ts">import NTP from "./NTP.svelte";
2: export let data;
[... 589 lines cut ...]
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

```
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Starting software access point
[   438][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   446][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   447][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
[   457][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[   461][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[   473][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[  1136][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[  1137][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_S[O 1138][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[  1143][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
Starting captive portal on 192.168.4.1
```

EvEggelen, 2023-10-29:
I switched to the latest release v0.2.2. In this release I do not see this problem anymore. I propose to close this issue. When it pops-up again, I will post it again with details how to reproduce.

closed 2023-11-05 by theelims