theelims/ESP32-sveltekit #9: RMT failed to initilize

f-io, 2023-09-15:
**Hardware:** [Wemos Lolin ESP32S3](https://www.wemos.cc/en/latest/s3/s3.html) (N16R8) - more or less similar to esp32-s3-devkitc-1

Unmodified latest release (except for env) does not work with build in RGB LED.
LED_BUILTIN/RGB_BUILTIN 38 same as for the devkit.

```
E (31449) rmt: rmt_set_gpio(526): RMT GPIO ERROR
E (31450) rmt: rmt_config(686): set gpio for RMT driver failed
[ 31353][D][esp32-hal-rmt.c:615] rmtInit():  -- TX RMT - CH 1 - 1 RAM Blocks - pin 245
[ 31359][E][esp32-hal-rmt.c:631] rmtInit(): RMT failed to initilize.
[ 31365][E][esp32-hal-rgb-led.c:18] neopixelWrite(): RGB LED driver initialization failed!
```

theelims, 2023-09-15:
Can you please share your platformio.ini?

theelims, 2023-09-15:
The issue with your board is, that it uses a Neopixel RGB instead of a normal LED. Your board is not compatible with the lightstate example. But this shouldn't be an issue if you adjust the code for your own needs.

closed 2023-09-15 by theelims

f-io, 2023-09-15:
Sure, I just used [default](https://github.com/platformio/platform-espressif32/blob/master/boards/lolin_s3.json) settings for now, cause there is no variant (flash, psram ..). I'm new to platform.io, but overriding defaults is more common on boards with multiple variants? The [env] will do the rest. Everything else is working fine 👍 

```
default_envs = lolin_s3

...

[env:lolin_s3]
platform = espressif32
board = lolin_s3
```

f-io, 2023-09-15:
> The issue with your board is, that it uses a Neopixel RGB instead of a normal LED. Your board is not compatible with the lightstate example. But this shouldn't be an issue if you adjust the code for your own needs.

The esp32-s3-devkitc-1 uses the same Neopixel RGB and the compiler takes care of that.  
As you can see in the log it will automatically call neopixelWrite() insted of digitalWrite().

I also tried it by manually change `digitalWrite(LED_BUILTIN, ...) `to `neopixelWrite(LED_BUILTIN, R,G,B);` [Example](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/BlinkRGB/BlinkRGB.ino) and changed the default wakeup PIN 38 and disabled this feature.

Once I use another io it works fine, but the failed initialization of the remote control transceiver is strange.

f-io, 2023-09-15:
@theelims is it working for [env:esp32-s3-devkitc-1]? I don't have this board on hand, maybe I will order one, but in the end the only real difference between these boards is a the USB-UART bridge (CH340x vs CP2102N).

theelims, 2023-09-15:
Yes, you need to check the pins which the ones you have on your board and adjust accordingly.

theelims, 2023-09-16:
> @theelims is it working for [env:esp32-s3-devkitc-1]? I don't have this board on hand, maybe I will order one, but in the end the only real difference between these boards is a the USB-UART bridge (CH340x vs CP2102N).

I tested this on my ESP32-S3-devkit**M**-1 board. And I think it's the **M** that makes the difference. They have rerouted the RGB-LED on this board to GPIO48. 

But in the end it shouldn't matter, except for the demo. Because you gonna delete the lightstate part anyway and replace it with your own needs.

f-io, 2023-09-17:
Thanks for looking at this again and of cause sharing your work! I'm thinking of contributing later. 

GPIO binding is not the problem. Platform IO env will take care of any hardware variants. The pre-compiler will replace BUILTIN_LED in this case with 38.  ~/.platformio/packages/framework-arduinoespressif32/variants/lolin_s3/pins_arduino.h

You are right, this doesn't matter for my own implementation, but if you include an example it should run as expected? The strange thing, I can utilize the RBG LED within Arduino IDE, but can't bring it up to live within VS, thats why I asked if you checked it. I will try to find the culprit.

theelims, 2023-09-17:
There are so many dev boards out there, you can't expect the example to run smoothly with each and every one of them. I test the code with the three boards in the default platformio.ini and with them it works. If you use a non standard board you are responsible to make the required changes to make the code work. 

In your case this even seems to be an issue with the platformio board definition. So you should consider filing an issue with them.

f-io, 2023-09-21:
jfyi, found the culprit.
https://github.com/espressif/arduino-esp32/pull/8659