theelims/ESP32-sveltekit #12: Building for lolin s2 mini fails

besanur, 2023-11-21:
```
Building in release mode
Retrieving maximum program size .pio/build/lolin_s2_mini/firmware.elf
Checking size .pio/build/lolin_s2_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.8% (used 58196 bytes from 327680 bytes)
Flash: [===Error: The program size (1422998 bytes) is greater than maximum allowed (1310720 bytes)
=======]  108.6% (used 1422998 bytes from *** [checkprogsize] Explicit exit, status 1
1310720 bytes)
========================================================================= [FAILED] Took 7.29 seconds =========================================================================

Environment    Status    Duration
-------------  --------  ------------
lolin_s2_mini  FAILED    00:00:07.291
```

closed 2023-11-21 by besanur

theelims, 2023-11-23:
@besanur I tried to compile it for the lolin_s2_mini and it works:
```
Retrieving maximum program size .pio\build\lolin_s2_mini\firmware.elf
Checking size .pio\build\lolin_s2_mini\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.8% (used 58196 bytes from 327680 bytes)
Flash: [=======   ]  72.4% (used 1422914 bytes from 1966080 bytes)
Building .pio\build\lolin_s2_mini\firmware.bin
esptool.py v4.5.1
Creating esp32s2 image...
Merged 2 ELF sections
Successfully created esp32s2 image.
```
I used this snipped in platformio.ini:
```
[env:lolin_s2_mini]
board = lolin_s2_mini
board_build.mcu = esp32s2
; Uncomment min_spiffs.csv setting if using PROGMEM_WWW with ESP32
board_build.partitions = min_spiffs.csv
```

I don't have a S2 board, so I can't test upload and coding. Also there might some specific build flags required for the USB functionality of the chip. 

Judging by the error you got you might have missed `board_build.partitions = min_spiffs.csv`.

besanur, 2023-11-23:
> @besanur I tried to compile it for the lolin_s2_mini and it works:
> ```
> Retrieving maximum program size .pio\build\lolin_s2_mini\firmware.elf
> Checking size .pio\build\lolin_s2_mini\firmware.elf [...]

Thank you for your efforts. I closed this issue again because I managed to fix it afterwards. I had forgotten some configurations. Firstly, the min_spiffs.csv.