theelims/ESP32-sveltekit #14: FIrst build problem

brad4ever, 2023-12-05:
Hi, I would like to use your project but unfortunately I can't even get to the first compilation.
I should have followed your instructions but during compilation it gives me this error.

> *  Executing task in folder theelims-ESP32-sveltekit: C:\Users\bradi\.platformio\penv\Scripts\platformio.exe run 
> 
> Processing az-delivery-devkit-v4 (board: az-delivery-devkit-v4; framework: arduino; platform: espressif32)
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> Skipping build interface step for target(s): 
> SSL Cert Store: Parsing certificates from ssl_certs\DigiCert_Global_Root_CA.pem
> SSL Cert Store: Successfully added 1 certificates
> SSL Cert Store: Successfully added 1 certificates in total
> SSL Cert Store: Successfully created src\certs\x509_crt_bundle.bin
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
> PLATFORM: Espressif 32 (6.4.0) > AZ-Delivery ESP-32 Dev Kit C V4
> HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES: 
>  - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
>  - tool-esptoolpy @ 1.40501.0 (4.5.1) 
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ strict
> Found 39 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- ArduinoJson @ 6.21.3
> |-- AsyncTCP-esphome @ 2.0.1
> |-- AsyncMqttClient-esphome @ 0.8.6
> |-- framework
> Building in release mode
> Compiling .pio\build\az-delivery-devkit-v4\src\main.cpp.o
> Archiving .pio\build\az-delivery-devkit-v4\libb8c\libWiFi.a
> Archiving .pio\build\az-delivery-devkit-v4\lib51c\libESPmDNS.a
> Compiling .pio\build\az-delivery-devkit-v4\libb67\FS\FS.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\libb67\FS\vfs_api.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\AsyncEventSource.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\AsyncWebSocket.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\WebAuthentication.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\WebHandlers.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\WebRequest.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\WebResponses.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\lib264\ESPAsyncWebServer\WebServer.cpp.o
> Compiling .pio\build\az-delivery-devkit-v4\libde1\DNSServer\DNSServer.cpp.o
> In file included from src/main.cpp:15:
> lib/framework/ESP32SvelteKit.h:48:10: fatal error: WWWData.h: No such file or directory
> 
> *****************************************************************
> * Looking for WWWData.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:WWWData.h"
> * Web  > https://registry.platformio.org/search?q=header:WWWData.h
> *
> *****************************************************************
> 
>  #include <WWWData.h>
>           ^~~~~~~~~~~
> compilation terminated.
> Compiling .pio\build\az-delivery-devkit-v4\lib72c\WiFiClientSecure\WiFiClientSecure.cpp.o
> *** [.pio\build\az-delivery-devkit-v4\src\main.cpp.o] Error 1
> ================================================================================================================= [FAILED] Took 10.24 seconds =================================================================================================================
> 
> Environment            Status    Duration
> ---------------------  --------  ------------
> az-delivery-devkit-v4  FAILED    00:00:10.236
> ============================================================================================================ 1 failed, 0 succeeded in 00:00:10.236 ============================================================================================================ 
> 
>  *  The terminal process "C:\Users\bradi\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
>  *  Terminal will be reused by tasks, press any key to close it. 

If I remove the PROGMEM_WWW option it obviously continues, but I would like to try using the non-spiffs version.
I have a further compilation problem in lightstateservice.cpp on LED_BUILTIN because I use an azdelivery kit v.4 which does not have this definition; I will bypass it by adding an external LED.... but the problem relating to the WWWData.h , I honestly can't understand what's causing it... I seem to have done all the first setup steps correctly.

Thank you in advance for the support

theelims, 2023-12-05:
You must use 'upload' instead of 'build' from the platformio commands to trigger the front-end building script which creates 'wwwdata.h'.

brad4ever, 2023-12-06:
Perfect! 
Good thing I thought I had followed all the steps correctly.... 
Works like a charm now. 
Thank you for your support and thank you for this awesome works

closed 2023-12-17 by theelims