theelims/ESP32-sveltekit #16: What is the constraint on use with ESP8266?

warwickchapman, 2023-12-17:
ESP12-E/F have 4MB on memory like entry level ESP32. What are the other constraints against use with ESP8266?

I have 4 different models of hardware we produce and 1 will continue to use an ESP8266. 

I’m happy to do some work to help add support for ESP8266 with sufficient memory. Just need a steer.

theelims, 2023-12-17:
The ESP8266 is End-of-Life and won't be manufactured any longer. I suggest you start redesigning your hardware. You have maybe 2 more years until it ceases to exist. The main issue is, that it's API is largely incompatible with the ESP32 framework. And I see no point in justifying the effort supporting a second API for a chip that is basically EOL. The ESP-C3-12F should be pin-compatible. 

Having said that, the project originates from https://github.com/rjwats/esp8266-react which as the name implies supports the ESP8266. If you compare the lib/framework folders it should be easily doable to bring the support for this chip back. However, I decided to drop the support to ease the maintenance effort for the code base and solely focus on the ESP32 series. And even there the API starts to split up already.

closed 2023-12-17 by theelims

warwickchapman, 2023-12-18:
Ok thanks.  I've seen predictions of the ESP8266's EOL since 2016 but it's like a bad smell that doesn't go away.  I'm looking and replacing the ESP8266 on our boards with an ESP32 and we can take it from there.

theelims, 2023-12-18:
@warwickchapman I got my information from https://www.espressif.com/en/products/longevity-commitment . Personally I'm even moving all my new projects to the S3 chip to have some peace of mind on that matter.

warwickchapman, 2023-12-19:
Well, that’s as authoritative as it gets. Thank you.