theelims/ESP32-sveltekit #25: [ ] Enable MQTT - Check/Uncheck Problem

nadolny, 2024-03-12:
After updating to version 0.3.0,  I conducted some tests and noticed an issue. 
I will describe the conditions to reproduce it:

1)  Configured the WiFi network:
     WiFi icon in the status bar displayed correctly.

2) Configured MQTT:
    URI: mqtt://192.168.1.12:1883
    Clean Session? (enable or disable)
    MQTT Status: Connected

3)  Disabled the  check box "[  ]Enable MQTT" checkbox:
    Applied settings
    MQTT Status: MQTT Disable

4)  Enabled MQTT again:
    Checked the "[ X ] Enable MQTT" checkbox
    Applied settings
    MQTT Status: None
    WiFi icon in the status bar changed to WiFiOff
    MQTT stopped functioning
    On the WiFi Station screen, the status remained as "Status: Connected"
    WiFi continued to operate
    MQTT functionality was restored only after performing a RESTART on the System Status screen.

Steps 2, 3, 4 : These steps repeat after the restart.

I conducted this test under two conditions:
   a) SDK Version > ESP-IDF v4.4.6-dirty / Arduino v2.0.14
   b) SDK Version > ESP-IDF v4.4.4 / Arduino v2.0.9

nadolny, 2024-03-14:
Examining the issue, I performed a check in the code:

File: MqttSettingsService.cpp
   void MqttSettingsService::configureMqtt()

    _mqttClient.disconnect();

    changed to:

    if (_mqttClient.connected())
    {
        _mqttClient.disconnect();
    }

I noticed that running the command

_mqttClient.disconnect();

with MQTT already disconnected, the application loops would stop. 
With the application loops stopped, the application behavior matched the initial message posted here.

"I would like to know your opinion regarding what was reported here. Thank you. 
Tests are ongoing...

theelims, 2024-03-14:
I need to look into that. Could also be related to the inner workings of the MQTT library. Since I'm on vacation it might need a week or two until I can fix this.

closed 2024-04-01 by theelims