theelims/ESP32-sveltekit #29: Minimize number of parallel sockets

theelims, 2024-04-01:
Right now the Lightstate Demo uses 2 sockets for SSE and WS and need a 3rd socket to update the HTTP pages. PsychicHttp limits the available sockets to 5 crippling the number of devices available for concurrent connections.

1. Create a UnifiedWebSocketServerEndpoint to use only one socket per frontend connection
2. Merge the NotificationEvents (SSE) into this  UnifiedWebSocketServerEndpoint
3. Use this additionally as an endpoint for the lightstate demo to show both usecases

theelims, 2024-04-06:
@runeharlyk I was looking around your github repos and found your work on the sockets, as well your previous ESP32 svelte template. Maybe we can sit down and discuss a couple of things round these two. Especially I am interested in your experience with that single file vite plugin. Does this work for SvelteKit? Right now the frontend creates ~64 endpoints. Each endpoint consumes a few bytes in heap. Thus larger apps with a heavier frontend put a lot of pressure on the available heap and it would be preferable to reduced the number endpoints significantly.

runeharlyk, 2024-04-06:
Very cool, I would definitely want to discuss them. 

As for the single-file-plugin - It only works with single entry apps as the plugin uses the inlineDynamic option which is not supported by rollup if there are multiple entries (See this [issue](https://github.com/richardtallent/vite-plugin-singlefile/issues/51) ). That is the reason my simple template only uses svelte with svelte-router.
But maybe we can build a new plugin that get around this restriction.

theelims, 2024-04-06:
Would you be open for other communication channels like discord? My handle is the same as on github.

I would like to understand how you do you socket implementation. From my point of view it should replace SSE and be available as a further endpoint (with the same mechanic and API as the other endpoints) for custom services. 

The other topic seems to be tough nut to crack open: https://github.com/sveltejs/kit/issues/3882

runeharlyk, 2024-04-07:
I tried adding you, but it seem like i cant send you a friend request. Maybe you can try adding me. My handles also match.

theelims, 2024-04-07:
I did sent you a request and also activated that everybody can send my a friend request.

theelims, 2024-04-19:
This issue is closed and a Event Socket System based on a single WebSocket connection will be included in the next release.

PR #43 (2024-04-17): Singular socket connection
# Purpose
Closes #29 that states that the current setup requires up to 3 concurrent connections to the server per user, which limits the number of total users. This pr replaces `NotificationEvents` with `EventSocket `which can
- Broadcasts arbitrary data to all connected clients
- Emits events with payload
- Pushs toast notifications (like in the old EventNotifications.h) to all clients
- Accepts event data from the client and call the callback of the concerning stateful service
- Handles both websocket and SSE

This make it possible for a client to subscribe to a nu [... cut]
closed 2024-04-19 by theelims