theelims/ESP32-sveltekit #51: WIFI with spaces

Amerlander, 2024-05-01:
When I use a space in the WIFI Name I get lots of errors `missing terminating " character`

So the [example in the documentation](https://theelims.github.io/ESP32-sveltekit/buildprocess/#factory-settings:~:text=%2DD%20FACTORY_WIFI_SSID%3D%5C%22My%20Awesome%20WiFi%20Network%5C%22) does not work: `  -D FACTORY_WIFI_SSID=\"My Awesome WiFi Network\"`

When I escape the spaces like `  -D FACTORY_WIFI_SSID=\"My\ Awesome\ WiFi\ Network\"` it builds and uploads, but the device does not connect to my wifi.

```
Compiling .pio\build\esp32dev\src\LightMqttSettingsService.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\src\LightStateService.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
<command-line>: warning: missing terminating " character
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\libeef\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiAP.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiGeneric.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiSTA.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiServer.cpp.o
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libeef\WiFi\WiFiUdp.cpp.o
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\liba5d\WiFiClientSecure\WiFiClientSecure.cpp.o
Compiling .pio\build\esp32dev\liba5d\WiFiClientSecure\esp_crt_bundle.c.o
Compiling .pio\build\esp32dev\liba5d\WiFiClientSecure\ssl_client.cpp.o
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libced\PsychicMqttClient\PsychicMqttClient.cpp.o
<command-line>: warning: missing terminating " character
<command-line>: error: missing terminating " character
lib/framework/WiFiSettingsService.h:182:24: note: in expansion of macro 'FACTORY_WIFI_SSID'     
             if (String(FACTORY_WIFI_SSID).length() > 0)
                        ^~~~~~~~~~~~~~~~~
<command-line>: error: missing terminating " character
lib/framework/WiFiSettingsService.h:185:29: note: in expansion of macro 'FACTORY_WIFI_SSID'     
                     .ssid = FACTORY_WIFI_SSID,
                             ^~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32dev\libfdd\ESPmDNS\ESPmDNS.cpp.o
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\libb9b\UrlEncode\UrlEncode.cpp.o
<command-line>: warning: missing terminating " character
In file included from lib/framework/ESP32SvelteKit.h:41,
                 from src/main.cpp:15:
lib/framework/WiFiSettingsService.h: In static member function 'static StateUpdateResult WiFiSettings::update(ArduinoJson::V6215PB2::JsonObject&, WiFiSettings&)':
lib/framework/WiFiSettingsService.h:184:64: error: expected primary-expression before '{' token 
                 settings.wifiSettings.push_back(wifi_settings_t{
                                                                ^
Compiling .pio\build\esp32dev\libf8d\FS\FS.cpp.o
Archiving .pio\build\esp32dev\libeef\libWiFi.a
Compiling .pio\build\esp32dev\libf8d\FS\vfs_api.cpp.o
<command-line>: warning: missing terminating " character
<command-line>: warning: missing terminating " character
Compiling .pio\build\esp32dev\lib8df\PsychicHttp\ChunkPrinter.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
<command-line>: warning: missing terminating " character
================================== [FAILED] Took 8.57 seconds ==================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:08.570
============================ 1 failed, 0 succeeded in 00:00:08.570 ============================ 
```

theelims, 2024-05-01:
I'll update the docs with the escape characters for blanks. 

Does it connect after a factory reset? Or is it not connecting to SSID's with blanks at all?

theelims, 2024-05-01:
Thank you. I'll update the docs accordingly for the next release.

closed 2024-05-01 by theelims