theelims/ESP32-sveltekit #52: Vite Proxy does not work: »Error: read ECONNRESET«

Amerlander, 2024-05-01:
I try to use the vite proxy to work on the sveltekit code. The board is available at 192.168.1.190, but the vite proxy does not seem to work:

Vite proxy config:
```
proxy: {
	'/rest': {
		target: 'http://192.168.1.190',
		changeOrigin: true
	},
	'/ws': {
		target: 'ws://192.168.1.190',
		changeOrigin: true,
		ws: true
	}
}
```

The Vite Server shows this error:
[image]

The serial output shows, that there was a request:
[image]

The error message seems very general to me, so I do not really have an idea where to expect the issue. Anyone had a similar issue before or an idea what I could check?

I'm using the ESP32_Relay_X8 which is equipet with an ESP32-WROOM-32E.

theelims, 2024-05-01:
Have you restarted the vite dev server after you've changed the config file? I know this failure and it usually disappears with a restart: CTRL + C --> `npm run dev`

Amerlander, 2024-05-02:
I tested different Browsers and **it works in Firefox (HTTP and WS Proxy)**
In Edge the HTTP Proxy also works, but not the WS Proxy.
In Opera it did not worked at all.

WS were also not working in Opera on the device IP without proxy.

I disabled the Browser-Default add-blockers and all add blocking extension, but that did not changed anything.

closed 2024-05-02 by Amerlander

theelims, 2024-05-02:
Interesting. I tested in on Firefox and Chrome and it did work on both. But glad you got it figured out.