theelims/ESP32-sveltekit #60: EventSource's response has a MIME type ("text/html") that is not "text/event-stream".

dkaukov, 2024-08-18:
Hi, after forking template and bulding it with default settings. Im getting lot of
```
Attempting SSE connection.
demo/:1 EventSource's response has a MIME type ("text/html") that is not "text/event-stream". Aborting the connection.
```
It seems like `/events` endpoint responds with default html:
[image]

theelims, 2024-08-18:
This seems to be browser specific. It works on Firefox. Might be the naming "EventSource" is problematic and it mistakes it with the build in event source which uses server sent events instead of websocket. 

You could try renaming the svelte EventSource into something different. 

I'll look into that later today.

dkaukov, 2024-08-19:
Yeah, the request from Chrome, seems missing:
```
Sec-WebSocket-Version: 13
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: xy4LT/rAJ+VJ+D1LIAgGgQ==
Connection: keep-alive, Upgrade
Upgrade: websocket
```

atassumer, 2024-09-25:
it's try to connect websocket connection to /events, but it must be /ws/events, delete www* .h file and build interface

closed 2025-03-26 by theelims