theelims/ESP32-sveltekit #67: kernel panic when using wifi on ESP32 S3

thedemoncat, 2024-12-08:
On the new ESP32-S3 N16R8 controllers, when connected to wifi, I started getting this problem. Moreover, on the second core, the rtos task is running correctly at this moment.

I found a [description](https://github.com/espressif/esp-idf/issues/11325?ysclid=m4fce1up1b329256052) of the problem, but something could not be fixed on their recommendation. Have you ever encountered such a problem?

```
ELF file SHA256: 378477d97c3eae71

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4038544b  PS      : 0x00020c34  A0      : 0x82049bb2  A1      : 0x3fcf2e10  
A2      : 0x00020c23  A3      : 0xa5a5a5a5  A4      : 0x82049bb2  A5      : 0x00060c25  
A6      : 0xfffbfff0  A7      : 0x00000046  A8      : 0x3fc98558  A9      : 0x00000001  
A10     : 0x60000000  A11     : 0x00000001  A12     : 0x0000000a  A13     : 0x3fcf2dfc  
A14     : 0x00000001  A15     : 0x3fcf6070  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x40385448:0x3fcf2e10 0x42049baf:0x3fcf2e20 0x4204a1b5:0x3fcf2e40 0x420277ba:0x3fcf2e60 0x42027aee:0x3fcf2ed0 0x40377469:0x3fcf2f20 0x40376df8:0x3fcf2f40 0x00040022:0x3fcf3000 |<-CORRUPTED

  #0  0x40385448:0x3fcf2e10 in xthal_window_spill at ??:?
  #1  0x42049baf:0x3fcf2e20 in esp_core_dump_replace_sp at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/espcoredump/include_core_dump/port/xtensa/esp_core_dump_port_impl.h:67
      (inlined by) esp_core_dump_setup_stack at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/espcoredump/src/core_dump_common.c:86
      (inlined by) esp_core_dump_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/espcoredump/src/core_dump_common.c:158
  #2  0x4204a1b5:0x3fcf2e40 in esp_core_dump_to_flash at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/espcoredump/src/core_dump_flash.c:342
  #3  0x420277ba:0x3fcf2e60 in esp_panic_handler at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:352
  #4  0x42027aee:0x3fcf2ed0 in panic_handler at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/panic_handler.c:188
  #5  0x40377469:0x3fcf2f20 in panicHandler at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/panic_handler.c:213
  #6  0x40376df8:0x3fcf2f40 in xt_highint4 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/soc/esp32s3/highint_hdl.S:119
  #7  0x00040022:0x3fcf3000 in ?? ??:0
```

theelims, 2024-12-08:
If you're running the default config all ESP32-SvelteKit functions run on Core 0. This seems an issue with your application or the Arduino main loop. It's consuming all CPU ressources and hasn't any headroom for idle. Check your code and add delay() in all tasks.

I only see this when I mess things up in my application.

closed 2024-12-08 by theelims

thedemoncat, 2024-12-08:
Thank you very much!
I'll probably check it again. I didn't reproduce the problem on your last commit. Apparently, I got something wrong(