theelims/ESP32-sveltekit #69: Unable to Make HTTPS Requests - Connection Refused with Framework

besanur, 2024-12-30:
Hello,

I am experiencing an issue where I am unable to make HTTPS requests to a server when using this framework.

I have tested the same HTTPS request code in a separate, minimal project without the framework, and it works as expected. In both cases, I used the same root certificate for SSL/TLS verification.

However, when using the framework, the request consistently fails with the error: "connection refused".

This suggests that the issue may be related to how the framework handles networking or SSL connections.

**Steps to Reproduce:**
1. Use the framework and attempt to make an HTTPS request with a valid root certificate.
2. Observe the "connection refused" error.
3. Run the same code in a standalone project without the framework.
4. Observe that the request works as expected.

**Expected Behavior:**
The HTTPS request should succeed when using the framework, just as it does in the standalone project.

**Actual Behavior:**
The HTTPS request fails with a "connection refused" error when using the framework.

**Environment:**
- Framework version: latest
- ESP32 Board: lolin_s2_mini

If there are any specific configurations or dependencies required for the framework to handle HTTPS requests correctly, please let me know. I am happy to provide more details or debug further if needed.

Thank you for your support!

```cpp
void requestData() {
  HTTPClient httpClient;
  WiFiClientSecure *client = new WiFiClientSecure;

  client->setCACert(test_root_ca);

  String url = API_URL + String(11027);
  Serial.print("Requesting data from server: ");
  Serial.println(url);

  // Führe HTTP GET-Request aus
  if (httpClient.begin(*client, url)) {
    Serial.print("Requesting data for ");

    int httpResponseCode = httpClient.GET();

    if (httpResponseCode == HTTP_CODE_OK) {
      Serial.println("data successfuly fetched");

      String payload = httpClient.getString();
      JsonDocument doc; 
      DeserializationError error = deserializeJson(doc, payload);

      if (error) {
        Serial.println("Failed to parse JSON");
      } else {
        Serial.println("Successfully stored data");
      }
      doc.clear();
    } else {
      Serial.printf("Failure during HTTP-GET. HTTP-Statuscode: %d", httpResponseCode);
      Serial.println();
      Serial.printf("Fehler: %s\n", httpClient.errorToString(httpResponseCode).c_str());
    }
  }
}
```

theelims, 2024-12-30:
It should work. At least it uses WiFiClientSecure for the OTA feature to download the binary from Github. I need to look deeper into it.

besanur, 2025-01-01:
> It should work. At least it uses WiFiClientSecure for the OTA feature to download the binary from Github. I need to look deeper into it. 

I found the problem. My wemos s2 mini does not have enough memory. The device causes me too many problems 😅.I don't have that with an esp32s3 supermini 💪

closed 2025-01-01 by besanur