theelims/ESP32-sveltekit #71: UI continuously show connected / disconnected toasts on screen in some cases

gbisheimer, 2025-01-21:
In some cases, I see the following behavior:

[image]

When this happens, I see the same result either accessing the UI directly from the device nor using the dev environment in my local machine and using the proxy configured in vite.config.ts to the device.

Is something related to the WS connection, but couldn't check it out yet.

Have you seen this before?

ewowi, 2025-01-21:
Hi there,

I have the same in the following situation:

- enabled security feature (user login)
- on Safari (I think not on chrome, not 100% sure though)
- Might also be related to the browser cache policy used

What is your situation regarding these bullet points ?

besanur, 2025-01-21:
In the original code, the if condition checks whether the user's bearer token is not an empty string before calling the validateUser function. However, the WebSocket initialization and event listener setup occur outside the conditional block, meaning they are executed regardless of whether the user is authenticated.

The improved solution moves the WebSocket initialization and event listener setup inside the if block. This ensures that these actions only take place after successful user validation, preventing potential issues such as unauthorized access or unnecessary WebSocket connections when the user is not authenticated.

+layout.svelte
from

```javascript
onMount(async () => {
	if ($user.bearer_token !== '') {
		await validateUser($user);
	}
	const ws_token = $page.data.features.security ? '?access_token=' + $user.bearer_token : '';
	socket.init(
		`ws://${window.location.host}/ws/events${ws_token}`,
		$page.data.features.event_use_json
	);

	addEventListeners();
});
```
to

```javascript
onMount(async () => {
    if ($user.bearer_token !== '') {
        await validateUser($user);
        const ws_token = $page.data.features.security ? '?access_token=' + $user.bearer_token : '';
        socket.init(
            `ws://${window.location.host}/ws/events${ws_token}`,
            $page.data.features.event_use_json
        );

        addEventListeners();
    }
});
```

gbisheimer, 2025-01-21:
This helped with the popups, but broke something. For instance, the connection status icon is not correct. Reloading the page shows the correct icon again.

[image]

Also have to modify login.svelte file from this:
```javascript
	async function signInUser(data: SignInData) {
		try {
			const response = await fetch('/rest/signIn', {
				method: 'POST',
				headers: {
					'Content-Type': 'application/json'
				},
				body: JSON.stringify(data)
			});
			if (response.status === 200) {
				token = await response.json();
				user.init(token.access_token);
				let username = $user.username;
				notifications.success('User ' + username + ' signed in', 5000);
			} else {
				username = '';
				password = '';
				notifications.error('Wrong Username or Password!', 5000);
				loginFailed = true;
				setTimeout(() => {
					loginFailed = false;
				}, 1500);
			}
		} catch (error) {
			console.error('Error:', error);
		}
	}
```

to this:
```javascript
	async function signInUser(data: SignInData) {
		try {
			const response = await fetch('/rest/signIn', {
				method: 'POST',
				headers: {
					'Content-Type': 'application/json'
				},
				body: JSON.stringify(data)
			});
			if (response.status === 200) {
				token = await response.json();
				user.init(token.access_token);
				let username = $user.username;
				notifications.success('User ' + username + ' signed in', 5000);

				const ws_token = $page.data.features.security ? '?access_token=' + $user.bearer_token : '';
				socket.init(
					`ws://${window.location.host}/ws/events${ws_token}`,
					$page.data.features.event_use_json
				);
			} else {
				username = '';
				password = '';
				notifications.error('Wrong Username or Password!', 5000);
				loginFailed = true;
				setTimeout(() => {
					loginFailed = false;
				}, 1500);
			}
		} catch (error) {
			console.error('Error:', error);
		}
	}
```

Remember to add the imports on top:
```js
	import { socket } from '$lib/stores/socket';
	import { page } from '$app/stores';
 ```

After this changes I get this:

[image]

I'll send a PR with this changes

theelims, 2025-01-22:
I did see this effect during development and thought I had fixed it. @gbisheimer which browser are you using? 

@besanur and @gbisheimer the fixes you provided will solve your case, but break compatibility if the security feature is disabled (which is what most people actually use).

I would like to reproduce the effect and find a clean solution that serves all use cases.

theelims, 2025-01-22:
I pushed a fix to head. I postponed the websocket connection to the event system to after successful user validation or login. No need for the event system on the login page.

closed 2025-01-22 by theelims

gbisheimer, 2025-01-23:
> I did see this effect during development and thought I had fixed it. [@gbisheimer](https://github.com/gbisheimer) which browser are you using?
>
> [@besanur](https://github.com/besanur) and [...]

I'm using Google Chrome. I'll try the new fix tomorrow.

Thanks!