theelims/ESP32-sveltekit #82: Questions about ESP Web Tools and EMBED_WWW

thedemoncat, 2025-03-26:
Hi!

In the changes, you indicated that you added the ESP Web Tools firmware build script.
I looked at the script and I had some questions., 
1. Is it made only for ESP32 with 4MB flash memory? Is it possible to flash ESP32-S3 with 16MB of memory in the same way?
2. If you build ESP-sveltekit with the EMBED_WWW option disabled, will it be able to assemble the firmware into a single file and upload it via ESP Web Tools?

theelims, 2025-03-26:
The script draws the right parameters from the PIO environment. It should automatically adjust everything according to the build target. I tested it with a S3 with 8MB and it works.

I haven't tested with EMBED_WWW disabled. However, since it merges bootloader.bin, partition.bin and firmware.bin into one file it should contain all the flash LITTLEFS files as well.

thedemoncat, 2025-03-27:
Thank you so much for your reply! 
And if I may, in order not to open another issue, I wanted to ask one more question:
I noticed that an open tab in one browser (an esp32 socket connection is established) in some cases blocks the opening of a new tab, both in the same browser and on another device. If you close the initial tab, the others will load. Is this correct behavior?

theelims, 2025-03-27:
Yes, this is unfortunately a limitation of the underlying PsychicHttp and ESP-IDF libraries. I wanted to test the updated AsyncWebserver library again at some point in the future.

closed 2025-03-27 by theelims